littlefs-project/littlefs #1069: Dont pass test with RAM "block  device"

BacCM, 2025-02-05:
`
#include <stdio.h>
#include <string.h>
#include "lfs.h"

#define VRAM_PAGE_SIZE (4096) 
#define VRAM_BLOCK_PAGES_COUNT (128)
#define VRAM_BLOCKS_COUNT (2048)

uint8_t* g_vram = 0;

int vram_read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size)
{
    memcpy(buffer, g_vram + block * VRAM_PAGE_SIZE + off, size);
    return 0;    
}

int vram_prog(const struct lfs_config* c, lfs_block_t block, lfs_off_t off, const void* buffer, lfs_size_t size)
{
    memcpy(g_vram + block * VRAM_PAGE_SIZE + off, buffer, size);
    return LFS_ERR_OK;
}

int vram_erase(const struct lfs_config *c, lfs_block_t block)
{
    memset(g_vram + block * VRAM_PAGE_SIZE, 0xff, VRAM_PAGE_SIZE * VRAM_BLOCK_PAGES_COUNT);
    return LFS_ERR_OK;

}

int vram_sync(const struct lfs_config *c)
{
    return LFS_ERR_OK;
}

// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
    // block device operations
    .read  = vram_read,
    .prog  = vram_prog,
    .erase = vram_erase,
    .sync  = vram_sync,

    // block device configuration
    .read_size = VRAM_PAGE_SIZE,
    .prog_size = VRAM_PAGE_SIZE,
    .block_size = VRAM_PAGE_SIZE * VRAM_BLOCK_PAGES_COUNT,
    .block_count = VRAM_BLOCKS_COUNT,
    .cache_size = VRAM_PAGE_SIZE,
    .lookahead_size = VRAM_BLOCKS_COUNT,
    .block_cycles = 500,
    .name_max = 16
};

void fill_test_buff(uint8_t* p, size_t sz, size_t seed)
{
    for (size_t i = 0; i < sz; ++i) {
        p[i] = (uint8_t)(i + seed);
    }
}

#define TEST_CHUNK_SIZE (1024)    
#define TEST_FILE_LEN_CHUNKS (1000)
#define TEST_FILE_NAME ("avocado")

lfs_t g_lfs;

int main(void) 
{
    g_vram = (uint8_t*) malloc(VRAM_PAGE_SIZE * VRAM_BLOCK_PAGES_COUNT * VRAM_BLOCKS_COUNT);
    assert (g_vram);

    int e = lfs_format(&g_lfs, &cfg);
    e = lfs_mount(&g_lfs, &cfg);

    assert(e == 0);

    uint8_t test_buff[TEST_CHUNK_SIZE];
    uint8_t read_buff[TEST_CHUNK_SIZE];
    
    fill_test_buff(test_buff, TEST_CHUNK_SIZE, 100);
    lfs_file_t file = { 0 };
    e = lfs_file_open(&g_lfs, &file, TEST_FILE_NAME, LFS_O_WRONLY | LFS_O_CREAT | LFS_O_EXCL);
    assert(e == LFS_ERR_OK);
    for (size_t c = 0; c < TEST_FILE_LEN_CHUNKS; ++c) {
        if (c == 4) {
            e = lfs_file_write(&g_lfs, &file, test_buff, TEST_CHUNK_SIZE);
        }
        else {
            e = lfs_file_write(&g_lfs, &file, test_buff, TEST_CHUNK_SIZE);
        }
        assert(e == TEST_CHUNK_SIZE);
    }
    e = lfs_file_close(&g_lfs, &file);
    assert(e == LFS_ERR_OK);

    e = lfs_unmount(&g_lfs);
    assert(e == 0);

    memset(&g_lfs, 0, sizeof g_lfs);

    e = lfs_mount(&g_lfs, &cfg);
    assert(e == 0);

    memset(&file, 0, sizeof file);
    e = lfs_file_open(&g_lfs, &file, TEST_FILE_NAME, LFS_O_RDONLY);
    assert(e == LFS_ERR_OK);

    lfs_soff_t sz = lfs_file_size(&g_lfs, &file);
    size_t len = sz / TEST_CHUNK_SIZE;
    assert(len == TEST_FILE_LEN_CHUNKS);
    for (size_t i = 0; i < len; ++i) {
        e = lfs_file_read(&g_lfs, &file, read_buff, TEST_CHUNK_SIZE);
        int cmp_res = memcmp(test_buff, read_buff, TEST_CHUNK_SIZE);
        assert(cmp_res == 0);
    }
    e = lfs_file_close(&g_lfs, &file);
    assert(e == LFS_ERR_OK);

    return -1;
}
`
msvc 2022 
fail on compare 4-t chunk

BacCM, 2025-02-05:
i failed with  my implementation

geky, 2025-02-06:
Hi @BacCM, I think you want:

``` c
g_vram + (block * VRAM_PAGE_SIZE * VRAM_BLOCK_PAGES_COUNT) + off
```

For vram_read, vram_prog, and vram_erase.

BacCM, 2025-02-06:
> Hi [@BacCM](https://github.com/BacCM), I think you want:
> 
> g_vram + (block * VRAM_PAGE_SIZE * VRAM_BLOCK_PAGES_COUNT) + off
> For vram_read, vram_prog, and vram_erase.

Thanks, I already found it

bmcdonnell-fb, 2025-02-06:
@BacCM,

> Thanks, I already found it

So close your issue, instead of asking others to continue to work on a problem you've already solved?

closed 2025-02-07 by BacCM