littlefs-project/littlefs #1072: Assert on block < erase_size correct?

Noxet, 2025-02-14:
I am testing different scenarios for littleFS. Currently, I have an STM32 with a sector size of 32 kiB, and I am planning on using 2 sectors. I wrote a small test with smaller sector size just to manually check the output:
```
// variables used by the filesystem
lfs_t lfs;
lfs_filebd_t bd;

// configuration of the filesystem is provided by this struct
const struct lfs_config default_cfg = {
    // block device operations
    .context = &bd,
    .read  = &lfs_filebd_read,
    .prog  = &lfs_filebd_prog,
    .erase = &lfs_filebd_erase,
    .sync  = &lfs_filebd_sync,

    // block device configuration
    .read_size = 32, // from STM docs
    .prog_size = 32, // -||-
    .block_size = 512,
    .block_count = 4,
    .cache_size = 32,
    .lookahead_size = 16,
    .block_cycles = 500,
};

const struct lfs_filebd_config default_filecfg = {
    .read_size = 32,
    .prog_size = 32,
    .erase_size = 1024,
    .erase_count = 2,
};

void test()
{
    char filename[32];
    int err = 0;
    int count = 0;
    do
    {
        snprintf(filename, sizeof(filename), "%08d", count + 1);
        err = createWriteFile(filename, "deadbeef", 8);
        lfs_remove(&lfs, filename);
        count++;
        if (count > 1000) break;
    } while(err == LFS_ERR_OK);
}
```

When I run this, I get the following assert error:
```
lfs.c:2213:debug: Expanding superblock at rev 3005
filebd: /home/dev/littlefs/bd/lfs_filebd.c:73: lfs_filebd_read: Assertion `block < bd->cfg->erase_count' failed.
```

I have configured the block size to be smaller than the sector size, and the block count to be higher.
I am under the assumption that `block_size * block_count = erase_size * erase_count` but please correct me if this is wrong.

Shouldn't it be possible to have multiple smaller blocks insize an erase block?
I am using the filebd emulation, so it might just be an error here, or is this the expected behaviour "for real"?

geky, 2025-02-17:
Hi @Noxet, it looks like that assert is working as intended.

littlefs requires the `block_size` to be a multiple (i.e. greater than or equal to) of the `erase_size`. littlefs erases at the block level, which wouldn't work if it was smaller.

Though it's worth noting littlefs will inline small files that can fit in `cache_size` in the file's metadata, so small files don't need a full block. File metadata is stored in small logs which are only limited by `prog_size`, so that might be what you're thinking of if you're more familiar with logging filesystems.

Noxet, 2025-02-17:
I see.
So, it might not be a good idea to use littlefs in systems with large sectors, like STM in this case? where a sector is 32kiB, and I can only use 2 blocks

closed 2025-03-10 by Noxet

geky, 2025-03-13:
It can work if all of your files are inlineable. Which is often the case for small config/kv stores that internal flash is often used for.

If you want to store larger files, probably not.

This is some ongoing work to remove the RAM requirement of inline files, but it's still a work in progress. If that lands, there'd still probably be a significant storage cost for inline files, but even ~8KiB files should be, in theory, inlineable with 32KiB block sizes.