littlefs-project/littlefs #1075: Dependencies in bare metal application

joncutting, 2025-02-21:
Hi,

I am working on integrating littlefs into firmware for an STM32F4 series microcontroller which is bare metal (no HAL, just CMSIS) C++ and is backed by a SPI FRAM device. All memory including read/write cache, lookahead, and file buffers are statically allocated. Compile flags include 
LFS_NO_MALLOC, LFS_NO_ASSERT, and LFS_NO_DEBUG. A static library compiles fine using ST's gcc-backed toolchain but when attempting to link it to my application I get linker errors about undefined references to _close, _lseek, _read, _write, _fstat, _isatty, _sbrk, _exit, and _kill. I never had those errors before. I am not using glibc or any alternative like newlib. Maybe I missed something stupid on my side, just wondering if there is anything about littlefs that requires glibc?

Jonathan

geky, 2025-03-13:
Hi @joncutting, thanks for creating an issue. That's certainly curious.

littlefs does call a couple of stdlib functions: strlen, strchr, strspn, strcspn, memcpy, memset, memcmp, but I've yet to see a stdlib where these aren't standalone functions (maybe they have internal asserts pulling in code? It may be worth trying to compile the stdlib with -DNDEBUG).

Worst case, you could define LFS_CONFIG with a path to your own lfs_util.h file to override these (these should all get lfs_ prefixes at some point...).

I have some standalone implementations [here](https://github.com/littlefs-project/littlefs/blob/2b3fdffe4cee6473ef01133c815e4dc58e186040/lfs_util.h#L601) you're welcome to copy. I want to upstream these eventually.

Also, just to double check, did you also define LFS_NO_WARN and LFS_NO_ERROR? littlefs currently has an annoying number of these flags...

rhempel, 2025-05-20:
I might be able to shed some light on this - your bare metal system does need some external libraries to be able to link successfully. The functions you say are missing are part of libc - so you WILL need to provide a library to implement these functions.

_sbrk is only useful if there is a call to malloc() somewhere, the other functions might be called if you have any file operations, for example a debug trace that does an fprintf() to STDOUT.

What would help is a copy of the error message that the linker spits out - that will tell you which function needs to be linked with _read (for example)

I don't use the arm-none-eabi libraries, I use https://packages.ubuntu.com/jammy/picolibc-arm-none-eabi