littlefs-project/littlefs #1076: lfs_setattr on open file

bailyzintl, 2025-02-24:
Are updated custom attributes written to disk immediately even on an open file? If I want to update a files contents and custom attribute atomically is the only way the tmp file and rename route?

geky, 2025-03-13:
Ah! So this is what [`lfs_file_config.attrs`](https://github.com/littlefs-project/littlefs/blob/0494ce7169f06a734a7bd7585f49a9fa91fa7318/lfs.h#L351-L362) is for.

The tricky part is that file-attached attrs need to be back by RAM, so you need to call `lfs_file_opencfg` to provide per-attribute buffers.

But any attributes in `lfs_file_config.attrs` are atomically written to the file's metadata when you call `lfs_file_sync` or `lfs_file_close`.

The RAM requirement is the reason there is no `lfs_file_setattr` or `lfs_file_getattr`. And, as it sounds like you have noticed, `lfs_setattr` writes to the disk immediately and may not be powerloss safe.

rcolombanWis, 2025-09-22:
The issue I found with `lfs_file_config.attrs` is that the file metadata is always written on sync/close, even if nothing has changed.

It is because when the file is opened, the `LFS_F_DIRTY` flag is added when reading the attributes.

rcolombanWis, 2025-09-29:
It's not super clean but I ended up doing like your second option by clearing the LFS_F_DIRTY flag on open and setting it when changing the attribute.

I find the `lfs_attr` struct more convenient than a dedicated API, having a specific flag set by a macro should be fine.

But other users might prefer an API, so having both and configurable is a good idea.