littlefs-project/littlefs #1077: Some suggestions for documents

MouriNaruto, 2025-02-25:
I found maybe LittleFS's specification and design documents lacks some information when I tried to implement the lightweight read-only LittleFS implementation for some lightweight scenarios like boot from MBR, handler for archivers, and etc.

For example, we don't know the maximum LittleFS volume size for current LittleFS implementation. Maybe is a number which near the 2^64 bytes if block size and block count are 0xFFFFFFFF. But I don't think it's a correctly value in current implementation. So, I open an issue.

Current we know the maximum file size for current implementation from the header file should about 2GiB. But I think that should also be mentioned in the documents. (Maximum file name and attribute length should also be mentioned in the documents too.)

Kenji Mouri

geky, 2025-03-13:
It's worth mentioning littefs can be compiled readonly (-DLFS_READONLY), in case that covers your use case.

But you're right these limits could be better documented. One concern is these limits are not well tested, are tricky to test, and historically prone to [subtlety](https://bugs.java.com/bugdatabase/view_bug.do?bug_id=5045582), so I'm hesitant to document exact limits until we actually test said limits.

littlefs uses signed block addresses/offsets in several parts of the code to mux with negative error codes, so I think a safer limit would be 2^62 (2^31 blocks_count x 2^31 block_size). If this seems lackadaisical, keep in mind littlefs currently runs into _severe_ performance issues before this point. 2^62 bytes, i.e. 4612 PiBs, is way beyond what most users care about anyways.

MouriNaruto, 2025-06-23:
Thanks you and the information is helpful for me. I'm sorry I reply you so late.

> It's worth mentioning littefs can be compiled readonly (-DLFS_READONLY), in case that covers your use case.

Maybe it will not cover my scenario. Here are some reasons:

- I hope to make some infrastructures which can make us able to use littlefs as bootable floppy disk to boot my experimental toy lightweight OS kernel on retro PCs. So I need to discover a ultra lightweight readonly implementation which can fit into 512 bytes boot sector. (Of course, the toy OS kernel itself will use the standard littlefs implementation.) If this progress is smooth. I will open source my implementation because I think it will more people will be happy for that.
- I also want to introduce the littlefs with multi on-disk versions image readonly support in my open-source archiver tool. I hope I can have a simplified and unified handler implementation, it will be good for people (include me) to learn the changes of littlefs on-disk formats, and it's also good for me to adapt your future on-disk format versions in the future.

At last, thanks you for the information of the readonly mode binary size for littlefs v2 and littlefs v3 mentioned in https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2951978084. 

Kenji Mouri