littlefs-project/littlefs #1088: "During our regression testing, repeatedly powering off the ECU abruptly while LFS operations were in progress caused the filesystem to become unwritable."

SiddharthPatel-Eaton, 2025-03-28:
Hi,

We are encountering a peculiar issue with our LFS integration on Serial EEPROM. During our regression testing, repeatedly powering off the ECU abruptly while LFS operations such as writing or deleting files are in progress causes the filesystem to become unwritable after several cycles. Although we can still read most of the other files, attempting to write a new file results in an exception, as shown in the attached image. Any assistance in understanding the cause or any pointers would be greatly appreciated.

[image]

SiddharthPatel-Eaton, 2025-04-11:
So apparently what appears is, if a block that contains metadata is somehow overwritten then there are no checks on when the pointers get used and hence it results in this erroneous behavior, but for now once we corrected the logic to ensure block write is proper, we are able to overcome this issue. so closing the issue for now, but it can be a future improvement in how the pointers are checked before use in littlefs, specifically the next pointers.

closed 2025-04-11 by SiddharthPatel-Eaton

geky, 2025-05-01:
Sorry about the late response, glad you found the issue.

Yeah, so this is a difficult thing to protect against in littlefs's design. Metadata logs (`lfs_mdir_t`) are updated atomically where they live, so we can't store the relevant checksum in the parent's pointer.

The good news is that a planned feature, global checksums, should improve this. Though it's currently bubbling down the pipeline and will take some time to stabilize: https://github.com/littlefs-project/littlefs/commit/1c5adf71b38f8eb26e5a00798bfdb5eac6c20a74