littlefs-project/littlefs #1101: Potential Security Improvements

minxiao-arch, 2025-05-08:
Hello littlefs Maintainers,

I sincerely appreciate your hard work on this project.

In today's digital landscape, the security of open - source software (OSS) has become a pressing concern. The Open Source Security Foundation (OpenSSF), a sub - foundation of the Linux Foundation, has been dedicated to enhancing OSS security for many years.

One of the valuable tools developed by OpenSSF is Scorecard. It provides a set of security checkpoints for OSS projects. After analyzing our project with Scorecard, it has identified several areas where we can improve security:

1. Branch Protection
Enabling branch protection rules and mandatory code reviews can significantly reduce the risk of introducing vulnerabilities. The important branches should be protected because it should not be deleted or forced pushed by mistaken.
You can check it in the Settings - Branches page, You can click the Add branch ruleset or Add classic branch protection rule to protect one or more branches. Current maintenance state: Branch protection is not maximized on development and all release branches.

2. Static Application Security Testing (SAST)
Implementing SAST tools is crucial as it allows us to detect vulnerabilities at an early stage of the development cycle.
You can check it in the Settings - Code Security page. You can enable the Code scanning options.

3. Dependency Update Tool
Using a dependency update tool ensures that our project always utilizes the latest and most secure library versions. You can enable dependabot in the repository settings.
You can check it in the Settings - Code Security page. You can enable the Dependabot options.

4. Security Policy
It is highly recommended to define a comprehensive security policy (SECURITY.md) in the root directory. This policy should include guidelines for vulnerability reporting and vulnerability publishment.
You can do it in the Security page which will give you a template file, just put some key informations(such as Email address or Vulnerabilities submission link) in the SECURITY.md and commit it.

For detailed information on these checks, you can refer to the [OpenSSF Scorecard documentation](https://github.com/ossf/scorecard/blob/main/docs/checks.md#check-documentation)

I believe that addressing these security improvements will strengthen our project's security posture. What are your thoughts on implementing these changes?

geky, 2025-05-13:
Hi @minxiao-arch, thanks for creating an issue.

I realize this comes from a place of good intentions, but checklists like these aren't really useful for a minimally funded project like littlefs. Adding a SECURITY.md, for example, implies that there is a paid team or person that can take action on security vulnerabilities, when there really isn't. And the time spent trying to check these boxes is time not spent of fixing issue, security-related or otherwise.

---

> Enabling branch protection rules and mandatory code reviews can significantly reduce the risk of introducing vulnerabilities. The important branches should be protected because it should not be deleted or forced pushed by mistaken.

The neat part thing about git is that it's distributed. I probably have ~100 copies of littlefs lying around. So even if the main branch is deleted it's easy to restore. It's true it's annoying if there's a wrong force push, but I'm not really sure it really counts as a _security_ vulnerability...

> Implementing SAST tools is crucial as it allows us to detect vulnerabilities at an early stage of the development cycle.
You can check it in the Settings - Code Security page. You can enable the Code scanning options.

Regarding SAST, there is some neat work being explored around filesystem fuzzing (https://github.com/littlefs-project/littlefs/issues/1030), but I have not been able to look into merging due to other priorities.

Regarding Github's "Code scanning". Is this Copilot driven? I'm not sure I have the highest trust for GitHub driven features in its current form. It would be better to have owned tools in CI we can control and limit false positives.

> Using a dependency update tool ensures that our project always utilizes the latest and most secure library versions. You can enable dependabot in the repository settings.

littlefs's only dependency is on the compiler and the C stdlib. Which you provide.

From my perspective, Dependabot has mostly been used as an excuse by GitHub to force breaking changes in GHA. Though it may be more useful for other languages.

BenBE, 2025-05-15:
Having a `SECURITY.md` doesn't mean you have a paid team. It just means that you have declared a channel where people should reach out to in case there's something with security impact, that in their opinion warrants not being discussed publicly.

Writing something like "Report issues by mail to foo@bar.tld. No guaranteed response time." is totally fine and properly sets expectations.