littlefs-project/littlefs #1102: Docs Request: Better explain the cases where LFS can be expected to recover in a crash, vs. when it won't

parker-research, 2025-05-12:
It seems that a power reset while the filesystem is mounted is safe. However, if the crash occurs while the write is occurring, it seems that there's a risk of data corruption.

I'd love if these cases of when crashes occur could be expressed in a table, with the expected result and estimated certainty (e.g., are there any cases where it's sorta 50-50 if it's survive)?

geky, 2025-05-13:
Hi @parker-research, thanks for raising an issue.

Under what situation are you seeing data corruption? Can it be reproduced locally?

There is a risk of data _loss_ if you lose power mid-write. But the relevant file should revert to its previous state in this case. No data corruption or partial writes should occur if you go through littlefs's filesystem APIs.

If you're seeing data corruption it could be a filesystem bug, but the model of littlefs shouldn't allow it normally.

Daniella-Ho, 2025-05-22:
Hi Geky,

Thank you for your explanation.
I am experiencing an issue related to data corruption.
My board uses NAND flash, and if a power loss occurs during a write operation, there is a risk of uncorrectable bitflips on the NAND. After rebooting the board in such cases, littlefs encounters an uncorrectable bitflip error when calling the FTL's read operation in lfs_bd_sync, which causes lfs_bd_sync to return an error.
Do you have any suggestions on how to handle this situation?
Thank you.

geky, 2025-06-01:
Hi @Daniella-Ho,

If your bd discovers a bit-error you should return LFS_ERR_CORRUPT.

Unfortunately, littlefs does not ([yet](https://github.com/littlefs-project/littlefs/pull/1111)) provide error detection/correction, so if introduced outside of powerloss this usually just gets forwarded to the user.

But if it is caused by powerloss, we catch the error from `lfs_bd_read` (not sync) and treat it as a failed commit ([example](https://github.com/littlefs-project/littlefs/blob/16ceb6793449fa159f99aed4a766c2823f59cf3e/lfs.c#L1172-L1175)).

---

Erroring in`lfs_bd_sync` after a powerloss is more curious. We do treat LFS_ERR_CORRUPT as a failed commit, but this is usually caused by a bad block or bus errors. I suppose it could be caused by power loss, but only if you also ended up with an [FCRC collision](https://github.com/littlefs-project/littlefs/blob/master/SPEC.md#0x5ff-lfs_type_fcrc).

You mentioned an FTL. Unfortunately powerloss safety requires handling throughout the entire stack. Is it possible the FTL changes other bits, even temporarily, during prog/read/erase operations? This would risk a corrupt disk after powerloss. I would check the FTL's documentation to see if it says anything about behavior under powerloss.