littlefs-project/littlefs #1107: Appropriate configuration for writing to NOR Flash

hhuynh222, 2025-05-15:
I have 512Mb Nor Flash where I need to create at least 9 files, each of which have different write sizes ranging from 24 bytes to 7000 bytes. What would the optimal configuration be to achieve this? 

This is what I currently have: 

    .read_size = 256,
    .prog_size = 256,
    .block_size = 4096,
    .block_count = 16384,
    .block_cycles = 100,

    // Caching and performance optimization
    .cache_size = 4096,
    .lookahead_size = 2048, 

    .compact_thresh = 0,
    .read_buffer = READ_BUFFER,
    .prog_buffer = PROG_BUFFER,
    .lookahead_buffer = LOOKAHEAD_BUFFER,

Is this correct or should read_size and prog_size = 1?

geky, 2025-06-01:
Hi @hhuynh222, these questions are difficult to answer, because there are so many moving parts in a filesystem (application behavior, flash performance, bus quirks, hardware caching, etc).

At some point the best option is to try both and see which one performs better.

That being said, I would suggest starting with `read_size=prog_size=1`, as this minimizes prog padding and reduces the number of times you need to compact. You could also try `prog_size=1` and `read_size=256`, which may make the best use on in-flash buffers without sacrificing prog padding. But it's hard to know in a vacuum.

geky, 2025-06-03:
Also worth noting work is going on in v3 (https://github.com/littlefs-project/littlefs/pull/1111) that solves this sync-padding issue, but it's not stable yet so maybe not the most useful for you right now.

geky, 2025-06-06:
Hmmm, that is curious. It sounds like littlefs may be internally calling `lfs_fs_mkconsistent` to prepare the filesystem for writing. Is this after powerloss? You may be able to move this work elsewhere with `lfs_fs_mkconsistent`, but it should only happen once after mount.

Traversal can also occur as a part of block allocation, but with a 2048 byte lookahead buffer this should only happen every ~16K blocks.

geky, 2025-06-06:
Ah, is it possible the traversal is coming from the `lfs_fs_gc` call?

In its current form, littlefs's gc is naive in that it needs to traverse all metadata to check if certain work can be done. Specifically metadata compactions.

One option is calling `lfs_fs_gc` less frequently, either after n operations or on a real-world timer. Another option is disabling eager metadata compaction (`compact_thresh=-1`) until you see it become a bottleneck. With `compact_thresh=-1`, `lfs_fs_gc` should only traverse to populate the lookahead buffere once every ~16K blocks.

In theory littlefs could keep track of if the filesystem mutated since the last compaction scan, but this is currently not implemented. (it may also take multiple `lfs_fs_gc` calls before littlefs knows all blocks have been compacted, since compacting one block can cause another to become uncompacted)

Improving garbage collection is also a focus in v3, though I realize this doesn't help you at the moment.

hhuynh222, 2025-06-06:
I can give it a try. Will provide an update

geky, 2025-06-08:
> i removed the lfs_fs_gc for now, except after mounting and disabled compact thresh.

I just remembered that `lfs_fs_gc` will still traverse as soon as 1 block is allocated. So if you're still calling `lfs_fs_gc`, it may be doing more traversals than necessary. If you're not calling `lfs_fs_gc`, `gc_compact_thresh` is ignored.

In v2, `lfs_fs_gc` is eager to accomplish work, which unfortunately is sometimes counterproductive since it can only do full traversals.

> After creating/writing to the files about 15 times, i see that there are several traversal that sometimes occur while either trying to open the file and other times while closing/flushing the data and the time is large which causes large open and close file times.

This sounds like normal block allocation scans, but that would be unlikely if you haven't written >~16K blocks since the last mount. It could also be bookkeeping cause by mdir relocations (wear-leveling). Do you still see this with `block_cycles=-1`?

hhuynh222, 2025-06-08:
Just wondering could it also be that I have several files of different sizes. I dont have any directories created. Wondering if it would be better to create directories and sort files of similar size

hhuynh222, 2025-06-09:
I created different directories corresponding to file sizes. I changed block_cycles = -1 and on one of the directories, I wrote 8 files, it went through 2 traversal before opening the file to write.

geky, 2025-06-09:
If you run the program for many more iterations, do you eventually stop seeing traversals?

These may be caused by mdir splits, where littlefs decides a metadata block (which can contain metadata for multiple files) is too full and needs to be split into two metadata blocks. Over time these should stop as the metadata settles into a stable distribution.

Of course with `block_cycles=-1` you are no longer wear-leveling. But at least you can set it to a larger number `block_cycles=1000`, `block_cycles=10000`, depending on how much you want performance vs distributed wear.

> Just wondering could it also be that I have several files of different sizes. I dont have any directories created. Wondering if it would be better to create directories and sort files of similar size

That may allow better metadata density for small inlineable files. But I don't know if the affect would be very noticeable.

In theory, you give each file its own directory to avoid the mdir splits (dirs do not share mdirs), but this could be very wasteful as each file would use 4KiB at minimum.

hhuynh222, 2025-06-09:
It does go away but reappear again later on if I keep writing

geky, 2025-06-10:
Ah, that may be the traversal for block allocation.

You can't _delay_ the traversals any more. littlefs only traverses outside of `lfs_fs_gc` when it can no longer make progress. What `lfs_fs_gc` does is allow you to move the traversals forward, But the implementation is a bit naive.

One option if you have an RTOS/are multi-theaded, is you can yield in your block device's read function. That would allow other threads to continue while the filesystem is traversing. Though other filesystem operations will still be blocked if you have the threadsafe API enabled...

hhuynh222, 2025-06-10:
Unfortunately, RTOS/multithread is not available for me to use. I am using bare-metal, state-machine approach.

geky, 2025-06-10:
I did put together this library: [coru](https://github.com/geky/coru), to make it possible to yield outside of an RTOS. But adopting it may require significant work.

Outside of that or other stack manipulation, you're probably hitting the limits of what's possible with littlefs's current API. Making GC work efficient has been a priority of v3, but it's a significant enough rework of the internal API that it will probably always be out of scope for v2.

hhuynh222, 2025-06-10:
So it would need to be yielded inside of .read/.prog/.erase/.sync?

hhuynh222, 2025-06-10:
Tried implementing coru_create_inplace for STM32G4, chip but getting assertion error. I set size to 2048