littlefs-project/littlefs #1114: v3-alpha: Discussion

geky, 2025-06-23:
```
                  ####                                 
               ######## #####    #####                 
             ################# ########  ####          
       #### ### ############################## ####    
     #############-#### ##_#########################   
    ########'#########_\'|######_####################  
     ##### ##########_.. \ .'#_._#### #/#_##########   
   ################_    \ v .'_____#_.'.'_ ########### 
  ################_ "'. |  .""  _________ '"-##########
  ########"#####..--.  /    .-"'|  | |   '"-####'######
    #############    \     /  .---.|_|_    ########### 
     ###########     |    /  -|        |-   ########   
                     )    |  -|littlefs|-              
                     |    |  -|   v3   |-              
                     |    |   '--------'               
                     |   ||     ' ' '                  
       -------------/-/---\\----------------------     
```

# littlefs v3 roadmap is here: https://github.com/littlefs-project/littlefs/pull/1111

Unfortunately GitHub renders PR discussion very poorly these days ([gh#130618](https://github.com/orgs/community/discussions/130618), [gh#112603](https://github.com/orgs/community/discussions/112603)).

Feel free to use this thread for **high-level v3-wide discussion**.

For bugs, low-level questions, concerns, etc, feel free to create new issues with "v3" in the title. I will label relevant issues with the [v3 label](https://github.com/littlefs-project/littlefs/issues?q=is%3Aissue%20state%3Aopen%20label%3Av3) when I can.

---

Comments from the PR, let me know and apologies if I'm misrepresenting anything:

- https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918128540
  - @dpgeorge notes rdonly code size is a concern
  - https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918462415
    - @geky response, no numbers yet
  - https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2951978084
    - @geky response, initial rdonly results
- https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918685454
  - @Ryan-CW-Code notes simple kv API will be useful
  - https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918709988
    - @geky response, kv is a stretch goal 
  - https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918764361
    - @Ryan-CW-Code notes kv API would help migration from flashdb, and metadata compaction improvements are valuable
- https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2994832219
  - @MouriNaruto notes documentation on limits would be useful, concerns on supporting v2+v3, asks about `lfs3_migrate` and v3-prefix branch
  - https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2997686118
    - @geky response, unsure on `lfs3_migrate`, v2 compat uneconomical, `lfs3_*` prefixes adopted, leb128s interesting for limits and extensibility

geky, 2025-06-23:
@dpgeorge https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918128540

Fantastic!  Great to see this v3 work making it's way into the world.

The code size increase would be the biggest concern for MicroPython.  Any optimisations to get that down would be welcome.  In particular we put read-only littlefs2 inside a 32k bootloader.  Do you have any idea how big the read-only version of v3 is?

geky, 2025-06-23:
@geky https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918462415

@dpgeorge, I don't know quite yet. The non-default builds aren't fully reimplemented, but will update this thread when they are. Read-only is the most interesting but also the most time consuming.

The good news is that rbyds (and red-black trees) and B-trees are much more complicated on the write side than read side, so the impact should be less than the default build, but I'll wait to stick my foot in my mouth until after I have actual numbers.

geky, 2025-06-23:
@Ryan-CW-Code https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918685454

Looking forward to the Simple key-value APIs in v3

geky, 2025-06-23:
@geky https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2918709988

@Ryan-CW-Code, it's worth noting the simple key-value APIs are stretch goals. They may or may not make it into v3 depending on how long things take and funding.

Or to put it another way, stretch goals won't block the release of v3. But it would be convenient to get them in while breaking the API, to minimize future API breaks.

Though if they don't make it into v3, I would like to introduce them in future releases, if possible. Note most stretch and out-of-scope features are mainly API related unlikely to affect the on-disk format.

geky, 2025-06-23:
@MouriNaruto https://github.com/littlefs-project/littlefs/pull/1111#issuecomment-2994832219

It's really nice to see this v3 work. It seems improved a lot. I want to say some of my opinions:

- I hope littlefs v3 can test and document some information like maximum volume size and file size. (both in theoretical and recommended) It will help for littlefs users to know how to use littlefs better.
- I respect with v3 will have breaking changes and think it will make us have a better design. But it will have more challenges to support all (in current stage should be three) major on-disk versions of littlefs images in lightweight scenarios like some MCU devices which need to accept users external storages which format with littlefs.
- In the current stage, we know littlefs v2 implementation's little v1 implementation only have the v1 to v2 migration. I hope littlefs will have a separate version migration implementation source files instead of making inline in lfs*.h and lfs*.c. I think it will be useful for you when having more on-disk versions. (I think it's predictable. Maybe you will create v4 even newer on-disk formats before 2030.) Also, it will makes littlefs implementation more simplified. I hope side by side support of littlefs v3 like v1-prefix and v2-prefix branches will be the default mode, and it will be helpful for users to add littlefs more vNext on-disk format support versions in the future.

Kenji Mouri

geky, 2025-07-14:
@SVTechSupport that's a very interesting question.                              
                                                                                
Regarding timelines. I'm hesitant to nail down any specific dates as this is the only time the on-disk format will really be malleable -- I don't really want to make a v4 this decade. There's also several unknowns (funding, user feedback, discovered regressions, etc) that make planning tricky.          
                                                                                
At the moment my best guess is 2026. In part because of some unrelated things I am juggling.                                                                  
                                                                                
---                                                                             
                                                                                
As for MISRA compliance, I think that falls into the work that I probably wouldn't do for free, but would be happy to be paid for.                        
                                                                                
Some concerns:                                                                  
                                                                                
- MISRA, as a paid standard, is in a bit of conflict with open source development.                                                                  
                                                                                
  The spec not being publicly available creates some practical problems. I can't really expect users contributing patches to conform to something they don't have access to, for example.                                            
                                                                                
- Looking at it from a what's-best-for-littlefs perspective, I'm not sure MISRA compliance is the best use of engineering time at this stage.           
                                                                                
  Proving functionality through testing/benchmarking is currently the best ROI in terms of building confidence in the on-disk format. Ensuring a bug free codebase is somewhat humorously a secondary concern. (To be clear, I do want littlefs to be solid and bug-free, but everything requires time.)             
                                                                                
  Also MISRA compliance != bug-free codebase, but you probably already understand that nuance.                                                       
                                                                                
- I'm not very experienced with MISRA, so I will struggle to have a good estimate of the time investment required.                                     
                                                                                
What may make the most sense is a fork of littlefs specifically for MISRA compliance. This could live in its own repo and receive periodic updates from upstream littlefs.                                                         
                                                                                
That would solve concerns around standard access and style conflicts, though depending on how difficult MISRA compliance is, backporting upstream work may be quite involved.                                                              
                                                                                
Such a fork could also act as a sort of LTS version. Though ensuring MISRA compliance would slow down bug fixing, so maybe not.                            
                                                                                
---                                                                             
                                                                                
How to organize funding is another tricky question. If you're interested in outright paying for such work at an SWE rate let me know.                       
                                                                                
In a perfect world, such funding would be distributed over other users who find MISRA support valuable, but I'm not sure such tooling is available... I can make it a sponsorship goal but I'm not sure I can issues refunds if the goal is not met.

TLDR I need to do some more research into bug/feature bounty programs that might work with littlefs...

geky, 2025-09-29:
Just a quick littlefs3 update (this may be the wrong place for this, but oh well).

RE: code size: After some time away from the project, yeah, the code size increase is not great. I've poked around a bit for potential code savings, but I think this is just the inherent cost of littlefs3 (due to the new features and a more robust coding style).

This pushes littlefs3 closer to embedded Linux filesystems, at least if you ignore RAM (from some other experiments):

```
logging benchmark (NAND):
            code    ram
littlefs3  35776   3672
littlefs2  16880   2880
SPIFFS     25201   3588
Yaffs2     39383  87252
```

Open to any PRs/suggestions if anyone else wants to poke around at reducing the code size (I'm becoming increasingly convinced strict-aliasing was a mistake).

In theory, some of the newer features can be disabled to save code (binary search, random writes, rebalancing compactions, etc), but I think these will be out-of-scope until after release for reasons below.

If this breaks an existing constraint (MicroPython, @dpgeorge), the best option is probably stay on littlefs2 until some time after littlefs3 releases/matures. Hopefully not indefinitely, but that is an option.

---

Unfortunately, in planning out the next steps, I'm looking more at narrowing the scope, not increasing it. One thing that has worked well for littlefs2/1 is a small number of configurations, though I realize this probably frustrates some users.

The `LFS3_KVONLY` and `LFS_2BONLY` modes have already caused problems for refactoring work, and I'm not convinced all permutations are tested well, so I will be reverting these for now and downgrading them to stretch goals (still keeping `lfs3_get/size/set/remove`, these are at least easy to test).

Please expect no stretch goals to be completed. These will be moved to a wishlist of sorts after release.

Delaying stretch goals to post-release may mean more API-breaking releases in littlefs's future (not disk-breaking!), but I think prioritizing disk stabilization is going to be best for littlefs3. I attribute most of littlefs's success to user's feedback/contributions, but that only works when users can, you know, actually use the thing.

g-berthiaume, 2025-10-03:
This is very exciting.
Thanks for your efforts!

For me, the only thing missing from littleFS v2 is the options to use a file as a ring-buffer.
On memory constrain devices, it would be great to reserve a 25% of my flash for an append only ring-buffer for my logs.

The https://github.com/littlefs-project/littlefs/issues/1144 issue already started to discussed this topic, but as it is tagged with "question", I thought it would be valuable to mention it here.

An interesting starting point would be the [tniessen/littlefs-ringbuffer](https://github.com/tniessen/littlefs-ringbuffer) library that was created for this very reason.

g-berthiaume, 2025-10-07:
Thanks for you answer @geky .
I will take time to digest this idea, as I'm not familiar with the `fruncate` functions and it's implication. I typically use the "2 files and `rename` when full strategy" on Linux, but having something cleaner would be great.

If I may ask another question, I would be curious to know what's the plan for porting v3 (`erase`, `program`, `sync`, `read`). I can't find any information about it on https://github.com/littlefs-project/littlefs/pull/1111.

From my perspective, one of the key limitiation of littlefsV2 is the blocking nature of its port that ["colors"](https://journal.stuffwithstuff.com/2015/02/01/what-color-is-your-function/) the rest of the API. Therefore, you either block and weight (non ideal for reponse time) or you need to use a scheduler (RTOS or couroutine). If I'm not missing something, this implies that littleFS can't be used on bare-metal project.

Again, thanks for your work.

BenBE, 2025-10-23:
Do you think making LFSv3 aware of "ringbuffer-like" files would make sense? Meaning I can set a fixed upper limit for a file size and LFSv3 will fruncate the file in (configurable) chunks.

Example: A log buffer with fixed-sized log entries of 42 bytes each could be set to 21 KiB with 42 byte fruncate chunks and whenever a new log entry splits over, the oldest entry gets fruncated atomically with that write.

Possible limitation: write size is forced to a multiple of fruncation chunk size in that scenario, but that shouldn't be an issue for this kind of usage.

geky, 2025-10-23:
> Do you think making LFSv3 aware of "ringbuffer-like" files would make sense? Meaning I can set a fixed upper limit for a file size and LFSv3 will fruncate the file in (configurable) chunks.

Maybe if more users ask for it. But I'm not sure it really adds value over the two function calls.

Keep in mind these are usually surrounded by `open`+`close` calls, and if you want to "pop" from the ringbuffer you'd need to also `seek`+`read`. Internally we'd just call fruncate anyways, and that adds either more code cost to everyone or another compile-time option to test.

> the oldest entry gets fruncated atomically with that write.

It's worth noting littlefs's file-level snapshotting makes the `write`+`fruncate` atomic to any observer outside of the current file-handle.

rojer, 2025-11-03:
i'm a bit late to the party with this v3 thing, just read the roadmap - pretty excited about more efficient random writes!
what about appends though? are they covered by this, i.e. no more file rewrites on appends?
i hope so, and if yes, i think it's worth mentioning,a s it impacts such common cases as logging.

geky, 2025-11-14:
> what about appends though? are they covered by this, i.e. no more file rewrites on appends?

Yes! I've started referring to this as the ["sync-padding problem"](https://github.com/littlefs-project/littlefs/pull/1111#n-logging), and it's fixed in v3 by allowing files to stash unaligned writes in B-tree nodes directly.

Sync still has a cost, it needs to write-out any cached data and append a metadata commit, but it should be ~2 progs (pages) instead of ~1 block.

geky, 2026-02-20:
In other news: I will be at [USENIX FAST '26](https://www.usenix.org/conference/fast26) next week (Feb 23-26) with a poster+WiP covering littlefs's new B-trees.

I realize it's extremely short-notice, but if anyone is in the Santa-Clara area and wants to meet up, let me know!

If you're not, here's the poster: [littlefs-btree-poster-8.pdf](https://github.com/user-attachments/files/25435152/littlefs-btree-poster-8.pdf). A bit of a weird format for the web (DESIGN.md WIP), but I think it's a good high-level view of littlefs's new data-structure.

geky, 2026-03-12:
Of course, as soon as I run the benchmarks again I find a [performance regression](https://github.com/littlefs-project/littlefs/pull/1111/changes/3c2828180957b0a11c0fe3364e9e37f0729dd306)...

With that out of the way, here's some of the more interesting benchmark results so far:

<details open>
<summary><b>Methodology</b></summary>

\- branch: [littlefs-benchmarks/v3-p26-later](https://github.com/littlefs-project/littlefs-benchmarks/tree/v3-p26-later)
\- hash: [`490412d015134af4353ac318124d6fccd6326b4b`](https://github.com/littlefs-project/littlefs-benchmarks/commit/490412d015134af4353ac318124d6fccd6326b4b)
\- plots: [littlefs-benchmarks/v3-p26-later-plots](https://github.com/littlefs-project/littlefs-benchmarks/tree/v3-p26-later-plots)
\- commands (these take a while):

``` bash
PRECISE=1 make -f Makefile_later_l_n plot
PRECISE=1 make -f Makefile_later_l_g plot
```

Currently things are in a bit of a rough state, sorry about the mess.

#### Simulation

littlefs's bd simulator is getting pretty nifty.

One concern was how to model NAND flash. Most NAND flash _technically_ supports byte-level reads, but internally uses a large hardware buffer that likes larger read sizes. On w25n01gv, for example, the overhead of hardware buffer vs bus is surprisingly 50/50.

To try to fairly evaluate littlefs's small-read-heavy design, I've extended emubd/kiwibd with a set of read/prog/erase "width" and "byte" counters, where "width" is configurable to emulate hardware buffer overhead. This should fairly penalize small reads, while rewarding hardware alignment (which littlefs ignores).

The most common storage in this space is flash, so I've been focusing on NOR and NAND flash using timing estimates available in Windbond's datasheets (unaffiliated, they just have easily accessible datasheets):

- NOR flash - w25q64jv, [datasheet](https://www.winbond.com/resource-files/W25Q64JV%20RevM%2012242024%20Plus.pdf), [timing estimates](https://github.com/littlefs-project/littlefs-benchmarks/blob/490412d015134af4353ac318124d6fccd6326b4b/runners/bench_defines.h#L232)

- NAND flash - w25n01gv, [datasheet](https://www.winbond.com/resource-files/W25N01GV%20Rev%20R%20070323.pdf), [timing estimates](https://github.com/littlefs-project/littlefs-benchmarks/blob/490412d015134af4353ac318124d6fccd6326b4b/runners/bench_defines.h#L275)

Also if anyone knows a good source of timing information for SD/eMMC and NVRAM/FRAM, those would also be interesting to explore.

#### Benches

To evaluate littlefs, I've come up with 4 benchmarks that I think accurately model the microcontroller storage space. Open to feedback:

1. [**sequential**](https://github.com/littlefs-project/littlefs-benchmarks/blob/490412d015134af4353ac318124d6fccd6326b4b/benches/bench_p26_wt.toml#L40) - Writes $w$ to a file until it reaches $n$, truncates to zero, and repeats. Does not call `lfs3_file_sync`.

2. [**random**](https://github.com/littlefs-project/littlefs-benchmarks/blob/490412d015134af4353ac318124d6fccd6326b4b/benches/bench_p26_wt.toml#L200) - Writes $w$ to random locations in a file bounded by $n$ repeatedly. Does not call `lfs3_file_sync`.

3. [**logging**](https://github.com/littlefs-project/littlefs-benchmarks/blob/490412d015134af4353ac318124d6fccd6326b4b/benches/bench_p26_wt.toml#L518) - Logs $w$ to a file bounded by $n$ repeatedly, calls `lfs3_file_sync` after each write, and either fruncates via `lfs3_file_fruncate` or rotates logs via `lfs3_rename`.

   This is arguably the most important bench, modeling the common case of streaming sensor/etc data into a file. It also emphasizes `lfs3_file_sync` cost, which has been a problematic bottleneck for littlefs2.

4. [**many**](https://github.com/littlefs-project/littlefs-benchmarks/blob/490412d015134af4353ac318124d6fccd6326b4b/benches/bench_p26_wt.toml#L359) - Create $w$-sized files repeatedly, choosing a file-name such that the number of files never exceeds $`n/w`$.

   This models using littlefs as a cheap key-value store.

By default, benches were configured to use write size $`w=\text{64B}`$, target size $`n=\text{1MiB}`$, and disk size $`d=\text{128MiB}`$. After a warm up to flush any post-format schenanigans, they were each ran for 1 simulated hour, measuring the amount of data written to find the simulated write-throughput.

Blahblahblah, here's the results so far:

</details>

<details open>
<summary><b>NOR flash - various target sizes</b></summary>

[image]

</details>

<details open>
<summary><b>NOR flash - various geometry</b></summary>

[image]

</details>
[... 29 lines cut ...]
   This is surprising but great for littlefs's design. It suggests the sometimes-egregious padding in large metadata progs has a relatively small impact vs the total size of metadata during compaction.

5. At least the "fragmented" mode behaves as expected. This new mode is an interesting side-effect of littlefs3's write heuristics, in which you can tell littlefs3 to inline all data directly in the B-tree (if anyone has a better name for this, I'm all ears).

   The result is a pretty big jump in **random** write performance. But also a pretty big jump in storage overhead (~4x), and horrible **sequential**/**logging** performance.

   Still, could be a very interesting per-file mode for **random** write use cases (database?).

6. Moving on to the geometry results, the main goal of littlefs3 is to improve performance on large block sizes, and littlefs3 generally performs better with larger blocks than littlefs2.

[... cut ...]
7. As for read/prog size, it looks like these will continue to be config options worth fiddling around with for best performance.

   On NOR, with its small 256B prog-only hardware buffer, it seems the smaller the better.

   On NAND, with the huge 2KiB hardware buffer, larger reads generally result in better performance, but this seems to fall off before reaching the full buffer size.

   Note the larger read/prog sizes also need more RAM for the larger software buffers. This is not true for block size.

AlixANNERAUD, 2026-11-04:
Small question about the on-disk format before it gets stabilized.

Contiguous / mmap-able files (#692) have come up a few times (XIP code, assets, ML weights, lookup tables), and you mentioned there that what's really needed on disk is a "pinned" flag, with contiguity only mattering at allocation time. Since v3 is planning data redundancy and dedup through a virtual → physical block map, would a pinned, single-fragment file still be possible in that design, e.g. excluded from dedup/redundancy so its physical location never changes?

I'm not asking for the feature in v3.0, just wondering if it's worth keeping a bit of room for it in the format so it can be added later without another breaking change. I left some notes on the v2 side in #692 if that's useful.