littlefs-project/littlefs #1116: LFS_ERR_CORRUPT when writing a big file on a NAND memory

Tamanar, 2025-06-24:
I am working on the [w25n512g](https://www.mouser.com/datasheet/2/949/w25n512gv_rev_c_112118-2487896.pdf) NAND memory on an STM32WB55. With the characteristics as follows:
* Page size: 2,048 bytes
* Number of pages per block: 64
* number of blocks: 512

I have managed to make work the example given by little fs (So mount, create a file, reading, and writing)

But in my project, I need to have .wav files on the NAND memory. Because it can't be mounted on Windows, I've said to myself that I will the data of the .wav myself.

Here's my configuration
```C
static const struct lfs_config cfg = {
    // block device operations
    .read  = stmlfs_hal_read,
    .prog  = stmlfs_hal_prog,
    .erase = stmlfs_hal_erase,
    .sync  = stmlfs_hal_sync,

    // block device configuration
    .read_size = 512,
    .prog_size = 512,
    .block_size = W25N512G_BLOCK_SIZE,
    .block_count = W25N512G_TOTAL_BLOCKS,
    .cache_size = W25N512G_PAGE_SIZE,
    .lookahead_size = W25N512G_PAGE_SIZE,
    .block_cycles = 500,
};
```
Here's the function I use to read and write 
```C
int stmlfs_hal_read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void* buffer, lfs_size_t size)
{
	assert(block < c->block_count);
    assert(off + size <= c->block_size);

    W25N512G_ReadPage(block,off,buffer, size);

    return LFS_ERR_OK;
}

int stmlfs_hal_prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void* buffer, lfs_size_t size)
{
	assert(block < c->block_count);

    W25N512G_WritePage(block,off,buffer, size);

    return LFS_ERR_OK;
}

int stmlfs_hal_erase(const struct lfs_config *c, lfs_block_t block)
{
	assert(block < c->block_count);

    W25N512G_EraseBlock(block);

    return LFS_ERR_OK;
}

int stmlfs_hal_sync(const struct lfs_config *c)
{
    // No specific sync operation needed for W25N512G
    return LFS_ERR_OK;
}
```

```C
W25N512G_Result_t W25N512G_ReadPage(uint32_t page_addr, uint16_t column_addr, uint8_t *data, size_t read_len)
{
    if (page_addr + read_len > W25N512G_TOTAL_PAGES)
    {
        return W25N512G_ERROR;
    }
    int ret = W25N512G_LoadPageToBuffer(page_addr);
    if (ret != W25N512G_OK)
    {
        return ret;
    }
    ret = W25N512G_ReadFromBuffer(column_addr, data, read_len);
    return W25N512G_OK;
}

W25N512G_Result_t W25N512G_WritePage(uint32_t page_addr, uint16_t column_addr, const uint8_t *data, size_t size)
{
    if (page_addr >= W25N512G_TOTAL_PAGES || column_addr + size > W25N512G_PAGE_SIZE)
    {
        return W25N512G_ERROR;
    }

    // Load data to buffer
    if (W25N512G_LoadProgData(column_addr, data, size) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    // Program page from buffer
    if (W25N512G_ProgramExecute(page_addr) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    return W25N512G_OK;
}

W25N512G_Result_t W25N512G_EraseBlock(uint16_t block_addr)
{
    uint8_t cmd[4] = {W25N512G_CMD_BLOCK_ERASE,
                      0x00,  // Dummy byte
                      (uint8_t)((block_addr & 0xFF00) >> 8), (uint8_t)(block_addr & 0xC0)};

    // Enable write
    if (W25N512G_WriteEnable() != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }

    W25N512G_CS_Low();

    if (W25N512G_SPI_Transmit(cmd, 4) != HAL_OK)
    {
        W25N512G_CS_High();
        return W25N512G_ERROR;
    }

    W25N512G_CS_High();

    return W25N512G_WaitForReady(W25N512G_ERASE_TIMEOUT_MS);
}
```
And here's how I try to write

```C
    fsman->fres = lfs_mount(&fsman->lfs, &fsman->cfg);
    fsman->fres = lfs_mkdir(&fsman->lfs, "fr");
    fsman->fres = lfs_unmount(&fsman->lfs);  // Unmount the filesystem to ensure changes are saved
    fsman->fres = lfs_mount(&fsman->lfs, &fsman->cfg);
    fsman->fres = lfs_file_open(&fsman->lfs, &fsman->file, "/fr/001.wav", LFS_O_RDWR | LFS_O_CREAT);
    //sounddata is too big need to write it in chunks
    for (size_t i = 0; i < sizeof(sounddata_data); i += 512) {
        size_t chunk_size = (sizeof(sounddata_data) - i < 512) ? (sizeof(sounddata_data) - i) : 512;
        fsman->fres = lfs_file_write(&fsman->lfs, &fsman->file, &sounddata_data[i], chunk_size);
        if (fsman->fres < 0) {
            // Handle error
            fsman->fres = Close_File_LFS(fsman);
            return fsman->fres;
        }
    }
    fsman->fres = Close_File_LFS(fsman);
    
```

But every time I get LFS_ERR_CORRUPT when i = 3584.

I tried different conf I saw on issue #11, but none of them changed the issue.
sounddata_data is a const unsigned char sounddata_data[12844] (and it's the smallest .wav I have)

Did I miss something ?

Thanks for your help

Tamanar, 2025-06-24:
Here's the log :
```
fs_mount(0x2001d4ac, 0x2001d5b4 {.context=0, .read=0x80211d5, .prog=0x8021241, .erase=0x802128d, .sync=0x80212d1, .read_size=512, .prog_size=512, .block_size=131072, .block_count=512, .block_cycles=500, .cache_size=2048, .lookahead_size=2048, .read_buffer=0, .prog_buffer=0, .lookahead_buffer=0, .name_max=0, .file_max=0, .attr_max=0})<\n>

lfs_mount -> 0<\n>

lfs_file_open(0x2001d4ac, 0x2001d52c, "/fr/001.wav", 103)<\n>

lfs_file_open -> 0<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x8033ff8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x80341f8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x80343f8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x80345f8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x80347f8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x80349f8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x8034bf8, 512)<\n>

lfs_file_write -> 512<\n>

lfs_file_write(0x2001d4ac, 0x2001d52c, 0x8034df8, 512)<\n>

Corrupted dir pair at {0x0, 0x1}<\n>

```

Tamanar, 2025-06-25:
Something that I saw is that when I I write:
W25N512G_Result_t W25N512G_WritePage(uint32_t page_addr, uint16_t column_addr, const uint8_t *data, size_t size)
{
    if (page_addr >= W25N512G_TOTAL_PAGES || column_addr + size > W25N512G_PAGE_SIZE)
    {
        return W25N512G_ERROR;
    }
}
The column_addr (so the offset) equals 2048, and the size equals 512.  So, of course, I got an error because the page is 2048 bytes, so writing 512 bytes at the end of the page will automatically fail. I try to modify  stmlfs_hal_prog by adding LFS_ERR_CORRUPT to try to relocate to another block, but I still got the error...

geky, 2025-06-26:
This is difficult to debug without knowing how the W25N512G driver is implemented. But coincidentally I implemented a W25N512G driver myself recently.

The question is what units are page_addr and column_addr?

- littlefs's `block` = `W25N512G_BLOCK_SIZE` = 64*2048 bytes
- littlefs's `off` = 1 bytes
- W25N512G's page_addr = ?
- W25N512G's column_addr = ?

Note littlefs's `off` is 1 byte, regardless of prog_size and read_size. prog_size and read_size control the _alignment_ of `off`, but not the units. You may need to map it to the units of page_addr.

Guessing from `page_addr >= W25N512G_TOTAL_PAGES`, page_addr may be in units of pages (2048), so you may need to use `off / W25N512G_PAGE_SIZE`.

Guessing from `column_addr + size > W25N512G_PAGE_SIZE`, column_addr may be in units of bytes? So `off % W25N512G_PAGE_SIZE` may work.

---

A note for anyone else reading this: prog_size = 512 looks incorrect with page size = 2048, but the W25N512G supports reprogramming a page 4 times to allow `2048/4 => 512` byte "columns" (I've also seen "subpages").

Tamanar, 2025-07-02:
Yes, i try to change the write block function with this 
```C
W25N512G_Result_t W25N512G_WritePage(uint32_t page_addr, uint16_t offset, const uint8_t *data, size_t size)
{
    if (page_addr >= W25N512G_TOTAL_PAGES || offset + size > W25N512G_PAGE_SIZE)
    {
        return W25N512G_ERROR;
    }

    // Load data to buffer
    if (W25N512G_LoadProgData(offset, data, size) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    // Program page from buffer
    if (W25N512G_ProgramExecute(page_addr) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    return W25N512G_OK;
}
W25N512G_Result_t W25N512G_WriteBlock(uint32_t block_addr, uint16_t offset, const uint8_t *data, size_t size)
{
	uint32_t startpage=((block_addr*W25N512G_PAGE_SIZE)+offset);
	uint32_t bytesleft=0;
	uint32_t newoff=offset%W25N512G_PAGE_SIZE;
	uint32_t bufptr=0;
    
    
    while(size>0)
    {  
        bytesleft = size;
        if ((startpage/W25N512G_PAGE_SIZE) != ((startpage + size)/W25N512G_PAGE_SIZE))
        {
            bytesleft = W25N512G_PAGE_SIZE - (startpage%W25N512G_PAGE_SIZE);  // Adjust bytes left to fit in the current page
        }
        W25N512G_WritePage(startpage, newoff, &data[bufptr], bytesleft);
        size -= bytesleft;
        bufptr += bytesleft;
        newoff = bytesleft;  
    }
    return W25N512G_OK;
}
```

To be closer to what littlefs want.
The datasheet states that the memory has 512 erasable blocks, each with 64 pages, and each page contains 2048 bytes of possible data.
To make my code, I changed it to what littleFs wants. What I understand is that the block_address will be between 0 to 511. And the offset will be the address within a block. So it can be 0 to 64*2048. So, to write at the correct address,s I understand that I need to calculate the page address I need to write within a block given the offset.
Does it seem more correct to you like that? (I haven't managed to make it work like that)
If you have a snippet of code on how you implement the W25N512G driver, I would be glad to read it because the documentation is very poor on the internet.

geky, 2025-07-08:
> If you have a snippet of code on how you implement the W25N512G driver, I would be glad to read it because the documentation is very poor on the internet.

Ah, maybe "implemented a W25N512G driver myself recently" were the wrong words. I implemented a simulation of the W25N512G, so I'm not sure it's very helpful. I'm not talking to hardware.

> ``` c
> uint32_t startpage=((block_addr*W25N512G_PAGE_SIZE)+offset);
> ```

Sorry I missed where `block` was passed as `page_addr` earlier. I think that is incorrect. I think you want something like this:

``` c
W25N512G_Result_t W25N512G_WritePage(uint32_t block, uint16_t off, const uint8_t *data, size_t size)
{
    // W25N512G_TOTAL_BLOCKS = 512
    // W25N512G_BLOCK_SIZE = 64*2048
    if (block >= W25N512G_TOTAL_BLOCKS || column_addr + size > W25N512G_BLOCK_SIZE)
    {
        return W25N512G_ERROR;
    }

    // map blocks to bytes
    uint32_t block_addr = block*W25N512G_BLOCK_SIZE;
    uint32_t column_addr = block*W25N512G_BLOCK_SIZE + off;
    // align page_addr to W25N512G_PAGE_SIZE
    uint32_t page_addr = block*W25N512G_BLOCK_SIZE + (off - (off % W25N512G_PAGE_SIZE));

    // Load page to buffer
    int ret = W25N512G_LoadPageToBuffer(page_addr);
    if (ret != W25N512G_OK)
    {
        return ret;
    }
    // Write new data to buffer
    if (W25N512G_LoadProgData(column_addr, data, size) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    // Program page from buffer
    if (W25N512G_ProgramExecute(page_addr) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    return W25N512G_OK;
}
```
``` c
int stmlfs_hal_prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void* buffer, lfs_size_t size)
{
	assert(block < c->block_count);

    W25N512G_WritePage(block,off,buffer, size);

    return LFS_ERR_OK;
}
```

But I don't know exactly what `W25N512G_ProgramExecute` expects. The above assumes a byte address.

Note I added a load buffer before programming. I'm not sure this is correct, but may be needing to keep the contents of the other columns in the page.

geky, 2025-07-08:
Skimming the datasheet, it looks like the commands want just the truncated parts of the address.

[image]

So maybe this is what it wants?

``` c
W25N512G_Result_t W25N512G_WritePage(uint32_t block, uint16_t off, const uint8_t *data, size_t size)
{
    // W25N512G_TOTAL_BLOCKS = 512
    // W25N512G_BLOCK_SIZE = 64*2048
    if (block >= W25N512G_TOTAL_BLOCKS || column_addr + size > W25N512G_BLOCK_SIZE)
    {
        return W25N512G_ERROR;
    }

    // map blocks to bytes
    uint32_t block_addr = block*W25N512G_BLOCK_SIZE;
    uint32_t column_addr = block*W25N512G_BLOCK_SIZE + off;
    // align page_addr to W25N512G_PAGE_SIZE
    // W25N512G_PAGE_SIZE = 2048
    uint32_t page_addr = block*W25N512G_BLOCK_SIZE + (off - (off % W25N512G_PAGE_SIZE));

    // Load page to buffer
    int ret = W25N512G_LoadPageToBuffer(page_addr / W25N512G_PAGE_SIZE);
    if (ret != W25N512G_OK)
    {
        return ret;
    }
    // Write new data to buffer
    if (W25N512G_LoadProgData(column_addr % W25N512G_PAGE_SIZE, data, size) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    // Program page from buffer
    if (W25N512G_ProgramExecute(page_addr / W25N512G_PAGE_SIZE) != W25N512G_OK)
    {
        return W25N512G_ERROR;
    }
    return W25N512G_OK;
}
```