littlefs-project/littlefs #1120: LittleFS On-Disk Format Migration Paths

BenBE, 2025-07-04:
In a comment by @MouriNaruto:
- In the current stage, we know littlefs v2 implementation's little v1 implementation only have the v1 to v2 migration. I hope littlefs will have a separate version migration implementation source files instead of making inline in lfs*.h and lfs*.c. I think it will be useful for you when having more on-disk versions. (I think it's predictable. Maybe you will create v4 even newer on-disk formats before 2030.) Also, it will makes littlefs implementation more simplified. I hope side by side support of littlefs v3 like v1-prefix and v2-prefix branches will be the default mode, and it will be helpful for users to add littlefs more vNext on-disk format support versions in the future.

_Originally posted by @MouriNaruto in https://github.com/littlefs-project/littlefs/issues/1111#issuecomment-2994832219_

Which brought up the question for me, if there are any plans for
1. Stand-alone (offline) On-Disk Format migration
2. Stand-alone (online) On-Disk Format migration
3. A reasonable migration path (any->newest vs. incremental v1->v2->v3)
4. Co-existance migration, where a migration path works similar to the way BTRFS allows for migration from ext4->btrfs
5. Migration preferences
6. (Semi-)automatic online migration
7. Parallel support of different on-disk formats in parallel for reading+writing

Reading your documentation in #1111 the parallel support for v2+v3 is unlikely, thus a solution will likely revolve around (offline) step-migrations?

geky, 2025-07-08:
I went ahead and added a link to this as a stretch goal in https://github.com/littlefs-project/littlefs/pull/1111.

littlefs v2 included the optional `lfs_migrate` function for v1->v2 migration, but I'm not sure it received much use. It's a high risk, low value (long term) function, and my understanding is most users preferred to migrate at the application level. Application level migration is a bit safer and allows cross-filesystem migration.

That being said I'm happy to implement `lfs3_migrate` for v2->v3 migration if it's useful for users and there is time, but it's low priority vs getting v3 out the door. I'm also happy to merge a migrate implementation if someone else comes up with one.

---

Some technical notes:

- Unlike v2, in littlefs v3, data blocks contain only raw data, and pointers to data blocks can reference arbitrary slices of data. This makes littlefs v3 a sort of "universal migrator" that can migrate from any filesystem as long as you have the space for the new metadata blocks.

- littlefs's metadata is stored in metadata pairs (mdirs) with both blocks only used during metadata compaction. In _theory_, you can use the spare block to perform the migration in-place, but it doesn't work so well in practice. Changes to the size of metadata can cause the new metadata block to overflow, so you may still need to allocate new metadata blocks.

- The old filesystem tree -> mtree conversion may be tricky, but the new directory-ids (dids) are arbitrary, if you use the old dir's block address as the new did I think it should work out without needing recursion.

---

> 1. Stand-alone (offline) On-Disk Format migration

I think the theoretical `lfs3_migrate` v2->v3 function satisfies this.

> 2. Stand-alone (online) On-Disk Format migration

Given that most microcontrollers execute out of internal flash and littlefs doesn't really support [XIP](https://github.com/littlefs-project/littlefs/pull/692), I think this is out-of-scope?

> 3. A reasonable migration path (any->newest vs. incremental v1->v2->v3)

v1 -> v3 migration is also possible ("universal migrator" after all), but I don't think the set of users using v1 and set users who want to migrate to v3 overlap.

> 4. Co-existance migration, where a migration path works similar to the way BTRFS allows for migration from ext4->btrfs

That's a neat feature, I hadn't heard of btrfs supporting ext4 migration with rollback.

In theory all you need is a mechanism to tell littlefs v3 what blocks are used by the (frozen) v2 image so it doesn't allocate the same blocks. This may be possible with the bad-block tracking feature planned for the optional on-disk bmap.

> 5. Migration preferences

What preferences are you thinking of?

> 6. (Semi-)automatic online migration

My thoughts on this are the same as with `lfs3_format`. Mount errors can be used to automatically migrate on the user's side:

``` c
// try to mount littlefs v3
int err = lfs3_mount(&lfs, &cfg);
if (err) {
    // try to migrate littlefs v3 -> v2
    err = lfs3_migrate(&lfs, &cfg);
    if (err) {
        // could not migrate, format?
    }
}

// v3 should be mounted here
```

I think one of the lessons from the v2 -> v2.1 upgrade was that we should avoid automatically upgrading irreversible disk formats without user action.

> 7. Parallel support of different on-disk formats in parallel for reading+writing

I think the best option here is to include both versions of the codebase. There is some reusable code in the caching layers, but littlefs v2 and v3 look very different internally, so I'm not sure sharing code is worth the added complexity.

---

Open to feedback though.

BenBE, 2025-07-08:
Thank you for your detailed response.

Reading through it, it looks like the proposed migration path will likely boil down to the following:

1. Migrations should be applied incrementally
2. Migrations follow the on-disk format versions v1->v2->v2.1->v3->…
3. Migrations should(???) include all intermediate versions (no skipping v1->v3 directly)
4. Migrations are performed on an unmounted (offline) filesystem (no online migration while mounted)
5. Migrations with roll-back may be a stretch goal, but require some pre-requisites (e.g. bad block marking) in the target on-disk format version to work

I hope I summarized this correctly.

Regarding what I meant with "preferences": That was about the preferred way that users of littlefs are "expected" to realize filesystem migration when using LittleFS. Thus, as you pointed out, when an user wants to be compatible with multiple versions of on-disk formats of LittleFS, inevitably a future implementation would end up something like (exaggerated):

```
// try to mount littlefs v3 – assuming this is the most common path
int err = lfs3_mount(&lfs, &cfg);
if (err) {
    // try to migrate littlefs v2 -> v3
    err = lfs3_migrate(&lfs, &cfg);
    if (err) {
        // try to migrate littlefs v1 -> v2
        err = lfs2_migrate(&lfs, &cfg);
        if (err) {
            // could not migrate, format?
            err = lfs3_format(&lfs, &cfg);
        } else {
            // Reattempt migration from v2 -> v3
            err = lfs3_migrate(&lfs, &cfg);
        }
    }

    if(!err) {
        // try mounting again after migration
        err = lfs3_mount(&lfs, &cfg);
    }
}

// v3 should be mounted here, iff !err
```

So basically creating a stack of operations, where each failed operation is pushed onto for being retried in reverse order, once the first migration succeeded. And even with just considering the v1->v2->v3 path, this implementation will be a quite legthy one¹ … :)

NB: Technically `lfs_format` should not auto-mount the filesystem, but rather just initialize the on-disk structures (AKA offline operation).

I think providing an additional function `lfs3_migrate_full`, which implements the whole migration path² might be a good call to avoid errors from missing any steps on the user side.

¹I'm sure, the above code snippet is likely very incomplete and might lack some migration paths.
²Subject to some compile flags that allow specifying the supported source on-disk format versions³
³0 meaning `lfs3_migrate_full` is disabled, 1 meaning v1->v3, 2 meaning v2->v3, 3 meaning v1->v2->v3, -1 meaning every source version (effectively a bit set)