littlefs-project/littlefs #1127: tags parsed by the script are inconsistent

Ldd309, 2025-07-30:
LFS version: v2.5.1
 
When I checked the tags of the filesystem superblock, I found that two directories had the same ID, when I use scripts/readmdir.py, `python .\readmdir.py .\output.bin 4096 0 1`, I find that the id of the freeze directory and the exchange directory were different., but when I use `python .\readmdir.py .\output.bin 4096 0 1 -l` command, the printed result was consistent with my own calculation, the id of the freeze directory and the id of the exchange directory are both 2, I would like to ask why it causes this result, is there a mistake in my tag calculation method?

[image]

I've read SPEC.md and DESIGN.md and reviewed the LFS code. I have a few questions about the LFS mechanism.
1. When metadata pairs are compacted,are the IDs recalculated and reallocated?
2. After compact, if the space utilization of a metadata pair exceeds 50%, and a split is performed, each containing half the valid data, will subsequent commits write to the new metadata pair, and will the remaining space in the old metadata pair be unused?
3. For each directory metadata pair, the ID should be unique. If a metadata pair is split, does the ID of the new block start from scratch?

Thank you for your valuable time answering these questions.

geky, 2025-08-11:
Ah, so, `readmdir.py` and `readmdir.py -l` have two different views of the mdir.

With -l/--log, readmdir.py shows the raw log contents as they exist on disk. Without it, by default readmdir.py shows the replayed log, which is how littlefs views the metadata. littlefs has to replay the log everytime it accesses it, but for us humans it's usually a bit easier to view the replayed metadata.

One thing to keep in mind is littlefs organizes files and other metadata entries into a virtual sorted array. Which means sometimes files are inserted in the middle of the array instead of at the end, with each "create" tag representing the insert operation. What you're seeing here is "exchange" being inserted before "freeze". It appears as though "exchange" has the same id as "freeze", but when replayed, "freeze" is shifted over to the next id:

```
00000104: 40100400  create           1    0
00000108: 00200405  dir              1    5  65 76 65 6e 74           event

effective array: [superblock, event]
             id:      0         1

00000204: 40100800  create           2    0
00000208: 00200806  dir              2    6  66 72 65 65 7a 65        freeze

effective array: [superblock, event, freeze]
             id:      0         1       2

00000304: 40100800  create           2    0
00000308: 00200808  dir              2    6  65 78 63 68 61 6e 67 65  exchange

effective array: [superblock, event, exchange, freeze]
             id:      0         1        2        3
```

---

> 1. When metadata pairs are compacted,are the IDs recalculated and reallocated?

Nope. Ids stay the same if the commit that triggered the compaction does not change IDs.

However ids will change if the mdir splits, with the ids in the second mdir being shifted over to represent a new virtual array:

```
effective array: [superblock, event, exchange, freeze, other]
             id:      0         1        2        3      4

                                     |
                                   split
                                     v

effective array: [superblock, event]   [exchange, freeze, other]
             id:      0         1           0        1      2
```

Also note that ids can change based on unrelated file insertions/deletions in the virtual array.

> 2. After compact, if the space utilization of a metadata pair exceeds 50%, and a split is performed, each containing half the valid data, will subsequent commits write to the new metadata pair, and will the remaining space in the old metadata pair be unused?

After a split each mdir should end up with roughly half of the original ids, so it depends which id is being written to.

It's an error for an mdir to have no id, when deleting files littlefs should drop any mdirs that become empty. 

> 3. For each directory metadata pair, the ID should be unique. If a metadata pair is split, does the ID of the new block start from scratch?

Yes, ids are unique in each mdir, but not unique across mdirs. Hopefully the above examples help clarify how this works.

To uniquely reference a file across the entire filesystem, you need a combination of its mdir address (both block addresses) and the metadata id.

This has been a bit of a challenge for integrations into other systems that expect immutable POSIX-esque inode numbers. In littlefs3 (https://github.com/littlefs-project/littlefs/pull/1111) the plan is to move to a packed mtree id + mdir id representation that at least fits everything in 31-bits, but they will still change on unrelated file insertion/deletion.

Ldd309, 2025-08-15:
@geky Thank you for your detailed response.I learned more after reading lfs_dir_getslice this function, thank you again for your patience.