littlefs-project/littlefs #1132: lfs_fs_preporphans makes invalid assumtions on bool type

elupus, 2025-08-12:
In the function lfs_fs_preporphans, the value of lfs_gstate_hasorphans is left shifted 31 steps to set the high bit of the tag.
https://github.com/littlefs-project/littlefs/blob/8e251dd675da00342d45dac78b6f627f119aed03/lfs.c#L4895

While lfs_gstate_hasorphans is defined as bool, it's value is actually same as  lfs_tag_size, in other word 0 or any nonzero value.

https://github.com/littlefs-project/littlefs/blob/8e251dd675da00342d45dac78b6f627f119aed03/lfs.c#L420

This causes the high bit, to not be set at all. I've not investigated what issues this might cause. It was found while analyzing a weird clang-tidy warning:

> "The result of the left shift is undefined due to shifting '0' by '31', which is unrepresentable in the unsigned version of the return type 'uint32_t'" on the left shift.

geky, 2025-08-14:
Hi @elupus, thanks for creating a PR.

This has come up a couple times before (https://github.com/littlefs-project/littlefs/pull/772, https://github.com/littlefs-project/littlefs/pull/677), but I'm hesitant to change it as it would just hide other unintentional reliances on _Bool in the codebase.

lfs_init at least contains an assert to hopefully catch non-c99 conforming compilers:
https://github.com/littlefs-project/littlefs/blob/8e251dd675da00342d45dac78b6f627f119aed03/lfs.c#L4208-L4212

I'd be happy changing it if we actually tested with non-_Bool bool, but that raises the question what standard would we actually be testing for? c89 is high-effort low-value because of the amount of missing features.

elupus, 2025-08-14:
Would you be ok with "hasorphans ? 0x8000000u : 0u" left shifting a boolean value is rather weird and will be an integer promotion.

BenBE, 2025-08-14:
There'S WORDBOOL on Win32 …

Anyway, writing `!!cond ? true_val : false_val` ist most clear AND should even catch most older compilers.

Regarding the standard: Are there any (good) reasons to stop below C99? And C99 guarantees a proper boolean type, thus resolving the whole problem …